MicrosoftDocs/windows-driver-docs #2653: bcdedit /set documentation has no description of the `dtrace` value.

bgianfo, 2021-01-22:
The `dtrace` bcdedit value is missing from the `bcdedit /set` documentation.

For reference of the dtrace setting see: https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/dtrace#installing-dtrace-under-windows 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 714c21f8-cf12-3d98-f62a-cff0d326874e
* Version Independent ID: fc5eabc0-451e-1fe3-705b-d486c21cb334
* Content: [BCDEdit /set - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/bcdedit--set)
* Content Source: [windows-driver-docs-pr/devtest/bcdedit--set.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/bcdedit--set.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2021-01-24:
@bgianfo . Thanks for pointing out.i add that setting after reading  reference site

DOMARS, 2021-01-26:
Hi @bgianfo - I added a DTrace section, which now links to the DTrace topic. Thanks for the suggestion, and the interest in improving the Windows Driver Developer documentation.

closed 2021-01-26 by DOMARS