MicrosoftDocs/windows-driver-docs #2654: Visual Studio 2019 issue

thedinosoar, 2021-01-24:
There is an issue in the newer versions of visual studio where there is no project templates for any drivers, this is because WDK doesn't always properly install the extension for visual studio.

To solve this issue:

- Make sure WDK is properly installed
- Locate "WDK.vsix" (Mine was in C:\Program Files (x86)\Windows Kits\10\Vsix\VS2019)
- Run WDK.vsix to install the extension for visual studio

This solve my problem, but there aren't any resources to fix this problem

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 287160ad-f2a5-789a-2d7d-ff336514c11e
* Version Independent ID: a138162d-2b80-725b-212b-277bae8f177d
* Content: [Write a Universal Windows driver (UMDF 2) based on a template - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-umdf-driver-based-on-a-template)
* Content Source: [windows-driver-docs-pr/gettingstarted/writing-a-umdf-driver-based-on-a-template.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/writing-a-umdf-driver-based-on-a-template.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-01-26:
Hi @thedinosoar, thanks for opening this issue. Can you please specify the VS version and WDK version you downloaded when you ran into this problem?

aviviano, 2021-02-01:
Hi @thedinosoar, thanks again for letting us know about this issue. We added a note to our main Download the WDK page (https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk) on how to resolve the issue.

closed 2021-02-01 by aviviano