MicrosoftDocs/windows-driver-docs #2659: Insufficient Startup Memory for Virtual Machines

stryqx, 2021-01-27:
The majority of Stop 0x74 bugchecks that I see are virtual machines that have their startup memory set too low.
The resolution in all of these cases is to increase the startup memory to at least 1-2GB depending on Windows version and the number/type of boot-time drivers loaded.
Given that this page is the first returned in a Google search (fourth in Bing) for "STOP 0x74", it would be a good idea to include this scenario in the documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c5fdc587-37f6-7dc7-0f7d-2ddc189a1123
* Version Independent ID: 5b82fd28-1d78-0f87-7605-fc5309e6d1cd
* Content: [Bug Check 0x74 BAD_SYSTEM_CONFIG_INFO - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x74--bad-system-config-info)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x74--bad-system-config-info.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x74--bad-system-config-info.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-02-03:
@stryqx - Thanks for the input that we should add information about available storage space to Check 0x1A. It is always a good idea to make sure there is space available on an OS and that the file system is operational.  I added wording about these factors to the topic and a link to the Blue Screen Data topic that provides some additional information. Thanks for the input and good luck with your bug check work on a virtualized version of Windows.

closed 2021-02-03 by DOMARS

stryqx, 2021-02-04:
Please re-read my issue. It has nothing to do with storage capacity or file system integrity.
The root cause of my Stop 0x74 bugchecks was insufficient memory allocated to the VMs.