MicrosoftDocs/windows-driver-docs #2661: HV memory range fragmentation bug (cause by: dynamic memory)

vivianezw, 2021-01-27:
The HV dynamic memory option causes the physical memory ranges to become heavily fragmented after a while on a modern HV VM. (For example the "quick create" option in the HV manager creates such a Win10 VM showing this behavior.)

How to reproduce: 
* use the quick create button in HV manager (dynamic memory will be on by default)
* Start the VM and wait. Not sure how long.
* Download the MS Sysinternals Rammap tool und look at the physical memory ranges.

For an example rammap screenshot look here: https://github.com/Velocidex/WinPmem/issues/25

RAJU2529, 2021-01-28:
@vivianezw . Which article are you referring to?

vivianezw, 2021-01-28:
That's part of the problem: there is only https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2012-r2-and-2012/hh831766(v=ws.11), listed under "previous versions", and it does not seem to reflect the current state. The dynamic memory option is a switch in the Hyper-v manager on (any) modern Win10 or server. This feature is rather new (and it has bugs I want to report, any help appreciated).

tedhudek, 2021-01-28:
Hi @vivianezw, thanks for the feedback.  We just write the docs, and this sounds like a possible product bug. I found the following page that describes a bug bounty for Hyper-V bugs - you could try that: https://www.microsoft.com/en-us/msrc/bounty-hyper-v

Good luck!

closed 2021-01-28 by tedhudek