MicrosoftDocs/windows-driver-docs #2664: Hal initialization failed

lemmensc, 2021-01-31:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e2df4737-5000-820c-37fe-62c6166c40b7
* Version Independent ID: 4b3c8192-e763-519e-d0ed-5f40227a7e00
* Content: [Bug Check 0x5C HAL_INITIALIZATION_FAILED - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x5c--hal-initialization-failed?redirectedfrom=MSDN)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x5c--hal-initialization-failed.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x5c--hal-initialization-failed.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2021-02-01:
@lemmensc . This page for improvement and for IT administrators. So you can post your queries in answers.microsoft.com

DOMARS, 2021-02-01:
@lemmensc - As @RAJU2529 points out, this page is for input on the Windows developer driver docs. If you are having an issue with a stop code, you might try posting your question on https://docs.microsoft.com/en-us/answers/products/windows. Good luck with resolving your issue.

closed 2021-02-01 by DOMARS