MicrosoftDocs/windows-driver-docs #2665: FONTS vs HTML page formatting realization.

kholt891, 2021-02-01:
TEXT in font aware html page formats should not clip out information on your document pages. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e17b7b0-6443-0141-0223-1792b7fd5dd0
* Version Independent ID: 6e3e2d87-8ec2-88d2-4968-a74a83b0b946
* Content: [First steps for USB client driver development - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-started-with-usb-client-driver-development)
* Content Source: [windows-driver-docs-pr/usbcon/getting-started-with-usb-client-driver-development.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/getting-started-with-usb-client-driver-development.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2021-02-01:
@kholt891 . Thanks for pointing out. I willl try to fix

billlattimer, 2021-02-01:
@kholt891 I'm not seeing this behavior on Edge or Chrome; can you advise the browser and version you're using?

kholt891, 2021-02-01:
Font size in Chrome (current) as used in a web page zoom set by using
ctrl-(-) keys.
Object (text) is clipped in the table, about half way into the page.

This page:
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-started-with-usb-client-driver-development

kholt891, 2021-02-01:
Correction.
Using EDGE as web browser.

billlattimer, 2021-02-01:
@kholt891 thanks for calling this out and for the additional information. This article uses an HTML table vs. Markdown, making it unable to scale correctly with page zoom. I'm addressing this now and the page will be updated shortly. All tables in our Docs will be converted to Markdown over time.

closed 2021-02-01 by billlattimer