MicrosoftDocs/windows-driver-docs #2668: Such as the Enterprise Windows Driver Kit (EWDK)?

PeterGV, 2021-02-01:
Hi Don,

Nice, clear, step-by-step procedures here.  My only request is related to the sentence:

_Using a command line environment that is used for building driver source code, such as the Enterprise Windows Driver Kit (EWDK), navigate to the CodeQL tools folder where the repository was cloned._

All well and good... but... S'pose we're NOT using the EWDK?  S'pose we're building in good old VS?  I *think* that's the more common case, and that would be a better example to document.

It might also be nice to document how one could add running codeQL as a post-build step to their VS driver build -- I'm **assuming** you can do that... I haven't tried it yet myself.

Thanks!

Peter

P.S.  This is all quite aside from whether I think using or requiring the use of CodeQL at this time is a good idea.  Which I don't.  But that's an issue I'm actively addressing separately with the PM.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5660df59-18ee-441d-afc7-023da97a7da5
* Version Independent ID: 1e1c4d4d-7c4b-6af2-64f2-d51162a8992f
* Content: [CodeQL and the Static Tools Logo Test - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/static-tools-and-codeql)
* Content Source: [windows-driver-docs-pr/devtest/static-tools-and-codeql.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/static-tools-and-codeql.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-02-01:
@PeterGV - Thanks for this useful input. A local VS build is certainly a common way to build drivers. I just tested calling CodeQL as a post build event, and that seems to work fine. So I will add this as another option in the topic.

```
1>Run CodeQL Rule
1>">>> Running CodeQL <<<"
1>"Review Test results located in -> C:\codeql-home\databases\kmdf.sarif"
1>Running queries.
1>Compiling query plan for C:\codeql-home\codeql\cpp\ql\src\Likely Bugs\Underspecified Functions\TooFewArguments.ql.
1>[1/1] Found in cache: C:\codeql-home\codeql\cpp\ql\src\Likely Bugs\Underspecified Functions\TooFewArguments.ql.
1>Starting evaluation of codeql-cpp\Likely Bugs\Underspecified Functions\TooFewArguments.ql.
1>[1/1 eval 1.4s] Evaluation done; writing results to codeql-cpp\Likely Bugs\Underspecified Functions\TooFewArguments.bqrs.
1>Shutting down query evaluator.
1>Interpreting results.
```

PeterGV, 2021-02-01:
It also works in a VS command prompt window... I JUST tried that myself.

Apparently, there's actually some sort of VS Plugin that makes all this stuff "more integrated" as well...  I haven't even explored that yet.

For me, that's for another day.

Thanks, Don, for always being willing to listen to a request/suggestion.

P

DOMARS, 2021-02-04:
@PeterGV - Thanks for this input on Visual Studio. I added a new section with an example batch file and some directions on the adding a post build event to the release build of the echo driver. The last part of the batch file uses devenv.exe / Edit to load the generated SARIF JSON results file in Visual Studio. If the results section is blank, then the rules passed. There is a VS Marketplace plugin that parses SARIF OASIS standard files described in the CodeQL docs. 
We are always open to suggestions and ideas, thanks again for this input.

closed 2021-02-04 by DOMARS