MicrosoftDocs/windows-driver-docs #2669: Cannot find "Hardware submission page"

ofilm-randy, 2021-02-03:
Cannot find "Hardware submission page"

[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7135005c-e7aa-19a2-7cfe-5003696da66a
* Version Independent ID: dba3259f-cb98-c2ab-9c56-860a731310d9
* Content: [Hardware submissions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-certification-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-certification-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-certification-submissions.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2021-02-03:
@tedhudek . did you logged Microsoft partner page using your organization credentials..

ofilm-randy, 2021-02-04:
Sorry, I have found the problem. After I complete the hardware registration, I can already display the options.

tedhudek, 2021-02-04:
Glad you got it fixed!

closed 2021-02-04 by tedhudek

tedhudek, 2021-02-05:
BTW you can go directly to the Hardware submission page by going to https://partner.microsoft.com/en-us/dashboard/hardware/Search.