MicrosoftDocs/windows-driver-docs #2671: Is there a document giving the mapping from NT status codes to Windows error codes?

guyharris, 2021-02-08:
As a developer of libpcap, including the pcap-npf.c libpcap code that runs atop the Npcap driver and library, I sometimes have to look at the kernel-mode driver code and try to guess what the NT status codes returned by the driver turn into when the user-mode code in Npcap, which is called by libpcap, makes a Windows API call.

Is there a table that indicates how NT status codes in drivers are mapped to Windows error codes in calls to `ReadFile()`, `WriteFile()`, `DeviceIoControl()`, etc.?  If not, there should be.

rodwiddowson, 2021-02-08:
Unofficial but [this ](https://www.osr.com/blog/2020/04/23/ntstatus-to-win32-error-code-mappings/) might help

tedhudek, 2021-02-24:
Thanks @guyharris, great point. I added a link to the doc in question. It's currently visible on GitHub but will get to the live site eventually.

closed 2021-02-24 by tedhudek