MicrosoftDocs/windows-driver-docs #2675: Missing information on enabling WDF ruleclass. Since more drivers are written using WDF it would be good to explicitly state that the /standard switch or /ruleclasses 34 can be used to turn on WDF's runtime verifier checks

Shyamal-V, 2021-02-10:
Missing information on enabling WDF verifier checks using verifier.exe. Since more drivers are written using WDF and it would be good to explicitly state that the "/standard" switch or "/ruleclasses 34" can be used to turn on WDF's runtime verifier checks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 13ec469b-0fa3-5088-ed15-c09bcb4d7d3d
* Version Independent ID: c24695dc-e291-d08f-5939-14ceb1d461b8
* Content: [Driver Verifier Command Syntax - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/verifier-command-line)
* Content Source: [windows-driver-docs-pr/devtest/verifier-command-line.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/verifier-command-line.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

tedhudek, 2021-02-11:
https://github.com/MicrosoftDocs/windows-driver-docs/pull/2676 is done and addresses some of this. Remaining is that we still need to document `/ruleclass` on https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/verifier-command-line

DOMARS, 2021-02-12:
@Shyamal-V - Thanks for this issue to add this new option. I created a draft topic and sent a link in email for review.

DOMARS, 2021-02-12:
@Shyamal-V - Thanks again for this issue. The updates to the verifier command line topic are live. Let me know if you see anything else we can improve.

closed 2021-02-12 by DOMARS