MicrosoftDocs/windows-driver-docs #2683: Missing links

psiodmak, 2021-02-17:
Step 1. says "(...) follow the procedure described in Enabling the Debugging of the Initialization Code (...) follow the procedure described in Enabling the Service Application to Break Into the Debugger" with no links. Which articles is it refering to?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d975fe54-c88a-52d2-10e3-6033a75b149d
* Version Independent ID: be14e8ee-d0c3-60d9-b80f-141d819721e9
* Content: [Debugging the Service Application Automatically - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugging-the-service-application-automatically)
* Content Source: [windows-driver-docs-pr/debugger/debugging-the-service-application-automatically.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugging-the-service-application-automatically.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

psiodmak, 2021-02-17:
Found it. They should link to [Enabling the Debugging of the Initialization Code](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/preparing-to-debug-the-service-application#-enabling-the-debugging-of-the-initialization-code) and [Enabling the Service Application to Break Into the Debugger](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/preparing-to-debug-the-service-application#-enabling-the-service-application-to-break-into-the-debugger)

RAJU2529, 2021-02-18:
@psiodmak , Thanks for pointing out. And  Thanking you correct link found by you. I am adding those links in this article,

DOMARS, 2021-03-05:
@psiodmak - Thanks for seeing that these links were missing and providing a link to the correct topics. And thanks to @RAJU2529 for creating a PR with those links. Updates to the topic are live.

closed 2021-03-05 by DOMARS