MicrosoftDocs/windows-driver-docs #2684: Attestation signing and Signing a driver for internal distribution only

Stuv7CB, 2021-02-17:
Hello.

As it was mentioned in #2649 signing driver for internal distribution doesn't work for win2016 and later. Also, both types (attestation and for internal distribution) of signing requires submitting .cab file with a driver.

Does "Signing a driver for internal distribution only" mean attestation signing? Can you please clarify this in documentation?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-#signing-a-driver-for-internal-distribution-only)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

Stuv7CB, 2021-03-05:
@tedhudek Hello. I have seen that you deleted section in 86d7b04b51e131892a05bb8ad6dfce1fdecac9b3. Does it mean that there is no option "signing a driver for internal distribution"

tedhudek, 2021-03-08:
Hi @Stuv7CB thanks for the feedback. We removed this page because it is no longer relevant information.

closed 2021-03-08 by tedhudek