MicrosoftDocs/windows-driver-docs #2698: What's 'Maximum Number of Secondary STAs supported'?

guliteb, 2021-02-23:
Hi,
On Windows 10 preview build 21313, there is a new capabilities/attribute called: Maximum Number of Secondary STAs supported. Is it the same meaning as last value in this page 'The maximum number of supported simultaneous STA/WFD clients.'?

If not, where can I find the definition of the 'Maximum Number of Secondary STAs supported'?

Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3e977c01-a8a4-ae66-effb-dcbd76d6b3d2
* Version Independent ID: 1ebe4ba6-a3ac-aeab-2b51-ec3b73a51577
* Content: [WDI_TLV_VIRTUALIZATION_CAPABILITIES - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/wdi-tlv-virtualization-capabilities)
* Content Source: [windows-driver-docs-pr/network/wdi-tlv-virtualization-capabilities.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/wdi-tlv-virtualization-capabilities.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-06-17:
Hi @guliteb, thanks for opening this issue. The 'Maximum Number of Secondary STAs supported' field is used for a new feature that we are introducing in the upcoming Windows releases and is not part of the WDI_TLV_VIRTUALIZATION_CAPABILITIES settings. We can't share further details about the field right now. Would you like to be contacted when those details are publicly available?

guliteb, 2021-06-18:
Yes, that would be good. Thanks.

aviviano, 2021-06-18:
@guliteb Please provide an email where we can reach you. Thanks.

guliteb, 2021-06-19:
guliteb@hotmail.com

aviviano, 2021-06-21:
Thanks @guliteb. We'll contact you when these details are available.

closed 2021-06-21 by aviviano

guliteb, 2021-08-31:
Hi Amy, do you have any details available for sharing?

aviviano, 2021-08-31:
Hi @guliteb, the details are not available yet. I do have you on my contact list and I'll let you know once the info is public.