MicrosoftDocs/windows-driver-docs #2700: WHEAErrorInjectionMethods::GetErrorInjectionCapabilitiesRtn example is out of date/has compile errors

pastramionry, 2021-02-24:
I'm not an experienced dev on windows desktop, C++, or WMI but this example appears to be lacking. While trying to incorporate this example I ran into a variety of issues. I also noticed similar issues trying to execute the WHEAErrorInjectionMethods::InjectError  method (the instructions for which can be found here: [https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/injecting-a-hardware-error](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/injecting-a-hardware-error) ).  For example take these lines here:

```cpp
...
// Get the status from the output parameters object
Result =
  pOutParameters->Get(
    L"Status",
    0,
    &Parameter,
    NULL,
    NULL
    );
Status = Parameter.ulval;
VariantClear(Parameter);
...
```
While implementing this the compiler issued errors complaining that `ulval` does not exist on VARIANT. This is because it needs to be ul**V**al. In Addition, the `VariantClear` function needs the argument to be passed by reference. So: `VariantClear(&Parameter);`

Furthermore, after fixing the issues I still was unable to access this WMI method in a user mode application. I could not successfully execute this method in C++, Powershell, WMI Test Tool, or WMI Code Creator. Using the WMI Code Creator the following C# code was generated for this method:

```csharp
using System;
using System.Management;
using System.Windows.Forms;

namespace WMISample
{
    public class CallWMIMethod
    {
        public static void Main()
        {
            try
            {
                ManagementObject classInstance = 
                    new ManagementObject("root\\WMI", 
                    "WHEAErrorInjectionMethods.InstanceName='WHEA_WMI_PROVIDER0'",
                    null);

                // no method in-parameters to define

                // Execute the method and obtain the return values.
                ManagementBaseObject outParams = 
                    classInstance.InvokeMethod("GetErrorInjectionCapabilitiesRtn", null, null);

                // List outParams
                Console.WriteLine("Out parameters:");
                Console.WriteLine("Capabilities: " + outParams["Capabilities"]);
                Console.WriteLine("Status: " + outParams["Status"]);
            }
            catch(ManagementException err)
            {
                MessageBox.Show("An error occurred while trying to execute the WMI method: " + err.Message);
            }
        }
    }
}
```
This attempt, and all of my other attempts, failed with following error: "An error occurred while trying to execute the WMI method: Generic failure"

There is no further information in the WMI section in Event Viewer as to why this might be. I have always run the code under an administrator account with the application elevated to "run as administrator", so I do not believe this to be an issue with permissions. Even still, I gave full permissions to this namespace to all in the WMI security MMC snap-in just to be sure. My testing was done on a fresh VM install of Win 10 Enterprise 20H2 64-bit using the Hyper-V manager's "quick create" feature. I used Visual Studio 2019 community edition as my IDE of choice and have installed version 10.0.19041.0 of the Windows 10 SDK. Thank you for looking into this matter.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8cd0269e-e029-43da-969c-2f28b780630d
* Version Independent ID: 033808d5-9a4f-b339-0905-26f7d83b2f6c
* Content: [Getting Error Injection Capabilities - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/getting-error-injection-capabilities)
* Content Source: [windows-driver-docs-pr/whea/getting-error-injection-capabilities.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/getting-error-injection-capabilities.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-02-27:
Hi @pastramionry, thanks for reaching out.

I fixed the code typos in 29de9303c8 - thanks.

The most likely issue is not WMI, but rather that the EINJ WMI interfaces return a generic failure if EINJ is not implemented on the system - which is by far the most common situation unless you happen to be a firmware or hardware dev with access to non-production machines.

closed 2021-02-27 by tedhudek

pastramionry, 2021-02-27:
Thanks @tedhudek. Appreciate the fix and the info. Sorry if I came off too crass, was just frustrated, heh.

tedhudek, 2021-02-27:
No worries, didn't seem that way at all -- appreciate your taking the time to write it up!

tedhudek, 2021-03-01:
BTW @pastramionry I just added 5cf02f9d2d with a bit more info

pastramionry, 2021-03-02:
Sweet! A fine update. Thank you for including this insight into the documentation.