MicrosoftDocs/windows-driver-docs #2709: Download link is broken

KK911, 2021-02-27:
The download link for [WinDbg Preview](https://www.microsoft.com/store/apps/9pgjgd53tn86) is broken. I get a `Service Unavailable` error after a few minutes.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8ea1adb7-1680-611f-a605-f8f10c7e26d8
* Version Independent ID: 37a27cc0-a7e9-adc0-2ce8-d9eb13df9c21
* Content: [WinDbg Preview - Installation - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-install-preview)
* Content Source: [windows-driver-docs-pr/debugger/windbg-install-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-install-preview.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2021-02-27:
@KK911 . download link is working, it will redirect to Microsoft store windbg preview app link

RAJU2529, 2021-02-27:
@KK911 , tr below link
**https://www.microsoft.com/en-in/p/windbg-preview/9pgjgd53tn86?rtc=1&activetab=pivot:overviewtab**

DOMARS, 2021-03-05:
@KK911 - Looks like the download link is now working for me and @RAJU2529. It is possible there was a short outage, but I think things are working now. Good luck with your work and thanks for your interest in Windows Debugging.

closed 2021-03-05 by DOMARS