MicrosoftDocs/windows-driver-docs #2722: "Scaling The Desktop Image" table not rendered correctly

AndrewGaspar, 2021-03-07:
In this file: https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/scaling-the-desktop-image.md

There should supposedly be a table after this text "The following table shows the values that are saved in the database and the values that are actually set.", but the items in the table are not formatted correctly - it's just a long stream of text.

tedhudek, 2021-03-12:
Good to hear from you Andrew!  Hope things are going well in NM

lorihollasch, 2021-04-01:
Wow, those tables sure were mangled! Thanks for reporting this, @AndrewGaspar - the updated page is now live.

closed 2021-04-01 by lorihollasch

AndrewGaspar, 2021-04-01:
> Good to hear from you Andrew! Hope things are going well in NM

Things are going well. Just started a new job at Facebook last week working fully remotely. (this issue is unrelated 😛)

I hope things are going well for you, too!