MicrosoftDocs/windows-driver-docs #2724: The redirection for local address and local port

matrixinline, 2021-03-08:
[Enter feedback here]
 if the connection has localhost ie 127.0.07 and port 80 as original and I want redirect it to 127.0.0.1 port 5334, it works so the statement it does not support local host and port redirection is wrong.. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4a3e4808-0c6b-8848-6270-8f5e8b810223
* Version Independent ID: d38592f5-3ca4-3d2a-20fd-c8cfd28b58d6
* Content: [Using Bind or Connect Redirection - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/using-bind-or-connect-redirection)
* Content Source: [windows-driver-docs-pr/network/using-bind-or-connect-redirection.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/using-bind-or-connect-redirection.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-04-12:
Hi @matrixinline. Thanks for opening this issue. Modifying the local address (or source address) of an outbound connection is not supported. Did you change the local address or destination address?
We've updated the documentation to clarify this: Changing the local address and port of a flow is only supported in the bind-redirect layer. This is not supported in the connect-redirect layer.
Please let us know if you have any further questions about this.

aviviano, 2021-04-14:
@matrixinline I'm closing this issue now, but please feel free to open a new issue if you have further questions.
Thanks.

closed 2021-04-14 by aviviano