MicrosoftDocs/windows-driver-docs #2732: Commit a product submission undocumented required headers

Conky5, 2021-03-10:
It appears a `Content-Length: 0` must be set otherwise the portal returns an http 411.

Also `Content-Type: application/json` seems to need to be set or the portal returns an http 415.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 58d4fba4-34f5-f000-fd4a-1acf9de277d0
* Version Independent ID: 638d0b07-faaa-1529-a6d5-3cd7a6b87838
* Content: [Commit a product submission - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/commit-a-product-submission)
* Content Source: [windows-driver-docs-pr/dashboard/commit-a-product-submission.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/commit-a-product-submission.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-03-10:
Thanks for the feedback @Conky5.  I've never gone thru this UI myself - can you tell me where on the doc page you're suggesting adding that info?  Better yet, click the Content Source link above and spin up a PR

tedhudek, 2021-03-11:
Fixed in #2734 - thanks again!

closed 2021-03-11 by tedhudek