MicrosoftDocs/windows-driver-docs #2737: Are Iranian companies allowed to sign up in the WHQL?

alikiani85, 2021-03-14:
Hi, i could not find anything regarding this in the documentation in MSDN

are Iranian people and companies allowed to sign up in the WHQL to get their drivers signed? assuming they already have an EV certificate.

This is important and Microsoft really needs to clarify this, because in 2 months many Iranian software developing companies that develop windows applications will no longer be able to develop apps..

If it is in fact restricted, then is there anyway companies could get a license/authorization to use it? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2daf70ed-7391-5783-583b-fcd99f1c222a
* Version Independent ID: 679bd45d-d0fd-a192-4d26-0121594fa86a
* Content: [Register for the Hardware Program - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/register-for-the-hardware-program)
* Content Source: [windows-driver-docs-pr/dashboard/register-for-the-hardware-program.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/register-for-the-hardware-program.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-03-18:
Hi @alikiani85 - thanks for reaching out.  Sorry, but I'm not sure.  Because this is not a documentation issue, I'm going to close out this issue.  Best of luck!

closed 2021-03-18 by tedhudek

alikiani85, 2021-03-18:
> Hi @alikiani85 - thanks for reaching out. Sorry, but I'm not sure. Because this is not a documentation issue, I'm going to close out this issue. Best of luck!

But it needs to be specified in the documentation of Hardware program signing up page as part of the restrictions for signing up (if they are in fact banned)
Microsoft really needs to clarify this..

alikiani85, 2021-04-02:
> Hi @alikiani85 - thanks for reaching out. Sorry, but I'm not sure. Because this is not a documentation issue, I'm going to close out this issue. Best of luck!

Ted i didn't get any response from you. so where exactly should we email/contact in Microsoft regarding this issue? i could not find any email address for Microsoft regarding this. We are asking a very simple question, which is are Iranians allowed to sign up in the Microsoft hardware program or not..

tedhudek, 2021-04-02:
Sorry, I don't have any info on this.

alikiani85, 2021-04-02:
> Sorry, I don't have any info on this.

Can you please at least let us know who should we contact in Microsoft regarding this? i searched the entire website, and there is no email or anything for either WHQL support or Microsoft policy support. we are not even asking a complex question here, the answer to it should be really simple, its either a yes or no.