MicrosoftDocs/windows-driver-docs #2738: Window Performance Kit: The KMDF section doesn't work.

ian910297, 2021-03-15:
I wanna use WPA to analyze behavior of drivers and I followed the two tutorial to set my computer.  

1. [Using KMDF Verifier](https://docs.microsoft.com/zh-tw/windows-hardware/drivers/wdf/using-kmdf-verifier)  
2. [Using the Windows Performance Toolkit (WPT) with WDF](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-the-windows-performance-toolkit--wpt--with-wdf)  

The following is what I have done step by step, but it doesn't work.  
* I set VerifierOn to 0x1 at first and set EnhancedVerifierOptions to 0xf000000.  
* Then, reboot the computer.   
* Launch Windows Performance Recorder and select the column of WDF Drivers. Then, save the result into an ETL file.  
* Finally, use Windows Performance Analyzer to browse the result.  

I couldn't find the KMDF section in compute section.

tedhudek, 2021-03-16:
Hi @ian910297, thanks for the feedback.  Did you try watching the video in https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-the-windows-performance-toolkit--wpt--with-wdf#getting-started?

Are you running latest Windows 10?

ian910297, 2021-03-16:
Yeah, I already watched the video. I am not running on latest Windows 10 and my environment is Version 1909, OS build 18363.1440.

I found another problem. I tried to start WPR command mode. And I wanna see the details of WdfTraceLoggingProvider.

```
C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit>wpr.exe -profiledetails WdfTraceLoggingProvider

        The Logging Mode does not match with the profile passed.

        Error code: 0xc5584017
```

That means I do not set EnhancedVerifierOptions correctly? Or it's WPR's bug?

ian910297, 2021-03-16:
I already watched the video and followed all the steps. Could you reproduce the result in video? @tedhudek   

Is there anything else I need to set for environment? Maybe I missed something important?   
* Edit INF file, add registry `parameters\Wdf` manually and enable VerifierOn  
* Run WdfPerfEnhancedVerifier.cmd  
* Reboot  
* Start to record with WPR  
* Browse the result with WPA  

I am not running in latest Windows 10. My environment is Windows 10 (1909, OS build 18363.1440).

closed 2021-03-19 by ian910297