MicrosoftDocs/windows-driver-docs #2741: Update referenced version of Windows 10 SDK

lordjeb, 2021-03-18:
Version of the refreshed Windows 10 SDK downloadable in step 1.5 is actually 10.0.19041.685.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2021-03-26:
@lordjeb . Thanks for pointing out. I am fixing it now

tedhudek, 2021-03-26:
Fixed in #2749 

Thanks for the report @lordjeb!

closed 2021-03-26 by tedhudek