MicrosoftDocs/windows-driver-docs #2746: No link to IOCTL_STORAGE_REINITIALIZE_MEDIA

tim-oleksii, 2021-03-25:
Hello,

There is no link to IOCTL_STORAGE_REINITIALIZE_MEDIA in page https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-command-set-support

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 40f20f63-14d2-cac3-fd82-4a4352918189
* Version Independent ID: addc1d08-2a90-d363-40c0-c8f83fe26699
* Content: [StorNVMe command set support - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/stornvme-command-set-support)
* Content Source: [windows-driver-docs-pr/storage/stornvme-command-set-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/stornvme-command-set-support.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

RAJU2529, 2021-03-26:
@tim-oleksii . Thanks for pointing out. I am going to add link now.

RAJU2529, 2021-03-26:
@e0i  .  Please look into this PR #2748

lorihollasch, 2021-03-26:
Fixed in #2748 thanks to @RAJU2529 .

closed 2021-03-26 by lorihollasch