MicrosoftDocs/windows-driver-docs #2750: WPP_INIT_TRACING advice

chrontrol, 2021-03-27:
Samples (specifically OsrUsbFx2 for UDMF2) and templates (eg User Mode Driver (UMDF V2)) disagree on the advice to update the WPP_INIT_TRACING macro given on this page (https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/how-to-generate-a-umdf-driver-from-a-kmdf-driver) entitled "How to convert a KMDF driver to a UMDF 2 driver (and vice-versa)". Both adhere to the KMDF rendition "WPP_INIT_TRACING(DriverObject, RegistryPath)"

tedhudek, 2021-03-30:
Thanks @chrontrol for the sharp eye.  I've updated step 4 on https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/how-to-generate-a-umdf-driver-from-a-kmdf-driver with a clarification.

We appreciate the feedback :)

closed 2021-03-30 by tedhudek

chrontrol, 2021-04-09:
Dear sir - With apologies, but while I've got a living human on the line...

Why do I always get this error: "This driver is not digitally signed!" when I convert my KMDF driver (based on osrusbfx2, works fine on XP) to UMDF2 for W10 ?

???

I've gone back to basics using VS19, but even the simplest template, "User Mode Driver, USB (UMDF V2)" right out of the box, produces the same result. (The only change I made was VID/PID in the .inf)

Now going on 2 weeks, I'm out of ideas.  Thanks in advance for any help.

Jim Durham
ChronTrol Corporation
***@***.***
619-282-8686
800-854-1999 US toll-free

tedhudek, 2021-04-09:
I think you want to use the procedure described in https://docs.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option.

Best of luck!