MicrosoftDocs/windows-driver-docs #2751: The Visual Studio new project dialog no longer looks anything like this. Making this extremely unhelpful :)

justin-murtagh, 2021-04-03:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 287160ad-f2a5-789a-2d7d-ff336514c11e
* Version Independent ID: a138162d-2b80-725b-212b-277bae8f177d
* Content: [Write a Universal Windows driver (UMDF 2) based on a template - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-umdf-driver-based-on-a-template)
* Content Source: [windows-driver-docs-pr/gettingstarted/writing-a-umdf-driver-based-on-a-template.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/writing-a-umdf-driver-based-on-a-template.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2021-04-06:
Thanks @justin-murtagh for opening this issue. We are commited to updating this page in the near future. In the meantime, do you have any specific questions we can help with?

justin-murtagh, 2021-04-06:
I worked it out for myself (thank-you though :)). My suggestion would be replacing the first screenshot on this page with one from the current version of Visual Studio. Also adding to this page exactly which features of Visual Studio you need to select in the installer to follow along to be able to do this would be extremely helpful.

mhopkins-msft, 2021-05-13:
Hi @justin-murtagh,

Thanks for reporting this. I've updated the screenshots and steps to match VS 2019.

 ~ *Mark*

closed 2021-05-13 by mhopkins-msft