MicrosoftDocs/windows-driver-docs #2753: Lack of full documentation for parameters.

piotrj3, 2021-04-07:
Parameters are not full described. I had investigation where i got such result:
```
WHEA_UNCORRECTABLE_ERROR (124)
A fatal hardware error has occurred. Parameter 1 identifies the type of error
source that reported the error. Parameter 2 holds the address of the
WHEA_ERROR_RECORD structure that describes the error conditon. Try !errrec Address of the WHEA_ERROR_RECORD structure to get more details.
Arguments:
Arg1: 0000000000000010, Error Source Type
```

But there is no description what Parameter 1 = 10 means.
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 45b35e68-8c01-c14f-40b2-0eef5412b402
* Version Independent ID: 141f44a5-d816-60be-18c9-710aad6b1cbb
* Content: [Bug Check 0x124 WHEA_UNCORRECTABLE_ERROR - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x124---whea-uncorrectable-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x124---whea-uncorrectable-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x124---whea-uncorrectable-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-05-08:
@piotrj3 - Thanks for this input. It does look like there are additional error types. I will investigate.

DOMARS, 2021-08-19:
@piotrj3 - Thanks again for pointing out this gap in the docs. I updated the topic to include parameters 0xC to 0x10. My understanding is that parameter 1s are defined in [WHEA_ERROR_SOURCE_TYPE enumeration (ntddk.h)](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntddk/ne-ntddk-_whea_error_source_type). 

The dev team really emphasized to work on these bug checks, the use of the !errrec <addr> to display the WHEA_ERROR_RECORD structure using the address provided in Parameter 2. This is typically done, using a full dump file. Taking those structures apart will provide much more information then is recorded in the bug check parameters. 

Good luck with your WHEA debugging, and thanks again for this PR.

closed 2021-08-19 by DOMARS