MicrosoftDocs/windows-driver-docs #2763: Can't understand how to automate the requestedSignatures field gathering

effolkronium, 2021-04-16:
Hello
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-product

Can't understand for sure, how to fill up the Signaturesdata field in the
https://manage.devcenter.microsoft.com/v2.0/my/hardware/products
API.

On dashboard we have checkboxes with a latest signatures. But there is not API to retrieve the available signatures list with latest OS versions.

Is it OK to stay with the list of signatures for old windows versions ?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 24518ab0-bd10-1a04-f3a0-1ba754409841
* Version Independent ID: ad632005-a43b-8d81-c4ef-8730d59b4363
* Content: [Create a new product - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/create-a-new-product)
* Content Source: [windows-driver-docs-pr/dashboard/create-a-new-product.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/create-a-new-product.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

effolkronium, 2021-04-17:
Comment: I want driver to run for all windows versions from xp to latest win10 update

mhopkins-msft, 2021-05-04:
Hi @effolkronium,

Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.

You may have success posting your question in the following forum, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[Windows Hardware WDK and driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk)

Again, thanks for reaching out. Best of luck with this issue.

closed 2021-05-04 by mhopkins-msft