MicrosoftDocs/windows-driver-docs #2764: "Add the Pipeline Config file to the driver package" is not working in VS2019

wujoho, 2021-04-16:
Couldn't figure out how to do the below steps:
[](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver#add-the-pipeline-config-file-to-the-driver-package)
[image]

In my VS2019 it shows as below, and not sure how to do the above 4 steps in the right pane?
For example, how to "Navigate to the project directory" from the right pane?
Tried clicking the _<Edit...>_, but no directory navigation window pops up.
[image]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 227ceb38-396a-aa19-6304-be3df9b0ca08
* Version Independent ID: 2f34593f-aa20-384a-b1bb-c8a962ec4706
* Content: [Build a basic v4 printer driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver)
* Content Source: [windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/building-a-basic-v4-printer-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-09-14:
Hi @wujoho, are you still working on building a print driver? I was able to find a workaround for the issue you found with adding the pipeline config file to the driver package.  

**Add the Pipeline Config file to the driver package**

1. In the **Solution Explorer**, select and hold (or right-click) *MyV4PrintDriver* project, then select **Properties** from the popup menu.

1. In the **MyV4PrintDriver Property Pages** window, expand **Configuration Properties** in the left pane.

1. Expand **Driver Install**, then select **Package Files**.

1. In the right pane, select then click the <Edit...> row in the listed files.

    [image]

1. Replace the placeholder with the relative path of pipeline config file, for example, `../MyV4PrintDriver Render Filter/MyV4PrintDriverRenderFilter-PipelineConfig.xml`, then click **OK.**

    [image]

1. Reopen the **Package Files** pane as in steps 1 - 3 above and verify that the pipeline config file is listed. For example:

    [image]

I was able to successfully build the solution at this point, with the last line of output showing:

`========== Build: 2 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========`

I have not yet worked on the second issue you ran into in the **Configure the driver solution for debugging and deployment** section. I'll check into that next.

Please let me know if the above steps work in your solution. If so I'll update the topic with the new procedure.

barrygolden, 2021-09-15:
Updated the **Add the Pipeline Config file to the driver package** instructions.

closed 2021-09-15 by barrygolden