MicrosoftDocs/windows-driver-docs #2767: Setup for submitting device driver to Microsoft for digit signing

lixintest, 2021-04-17:
[Enter feedback here]

Following the documented procedures, at the last step, it pops up "Sorry, something went wrong, please try it again....". I retried a few times per day in last 2 weeks, it always end up the same error message. 
What is the exactly went wrong? How to make it right? Is it web site error, or else?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2daf70ed-7391-5783-583b-fcd99f1c222a
* Version Independent ID: 679bd45d-d0fd-a192-4d26-0121594fa86a
* Content: [Register for the Hardware Program - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/register-for-the-hardware-program)
* Content Source: [windows-driver-docs-pr/dashboard/register-for-the-hardware-program.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/register-for-the-hardware-program.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

lixintest, 2021-04-17:
[image]
This is the error message screenshot.

mhopkins-msft, 2021-04-26:
Hi @lixintest,

Thank you for reporting this. We're looking into the issue and will let you know when we have more info to share.

Thanks,
 ~ *Mark*

mhopkins-msft, 2021-04-28:
Hi @lixintest,

Is there any more information you can share? For example, can you copy the error information and tell me any more specific error details, without sharing any personal info or credentials, of course.

Thanks,
 ~ *Mark*

lixintest, 2021-04-28:
Hi Mark,

Thanks for responding to my issue.
Here is the link to mp4 that recorded every step as how I got into this problem.
Any help is highly appreciated!
https://drive.google.com/file/d/1KJUf1HcDDGav2rHa19_z2_W1qjWnNgcW/view?usp=drive_web
[https://lh6.googleusercontent.com/w6-MWsYPfz-Cdd-Sgb7ZqLZgtPBvSNu2h0JuklPh-UqwrbXAm_iCddKtnQL1UvFeOeg=w1200-h630-p]<https://drive.google.com/file/d/1KJUf1HcDDGav2rHa19_z2_W1qjWnNgcW/view?usp=drive_web>
HardwareRegistrationFailed-4.mp4<https://drive.google.com/file/d/1KJUf1HcDDGav2rHa19_z2_W1qjWnNgcW/view?usp=drive_web>
drive.google.com

Thanks,

Lixin Lu
----------------------------------------------------------------------
CEO/Founder
TriagingX Inc.
6050 Hellyer Ave, Suite 150-6, San Jose, CA 95138
1.408.568.7372

www.triagingx.com<http://www.triagingx.com/>

Provide the best automated endpoint forensic investigation solution

mhopkins-msft, 2021-05-04:
Hi  

Sorry that this sat. 

For specific questions, Hardware Dev Center Support can investigate. To contact support and open a Hardware Dev Center support case for investigation, **first**, sign into your hardware dev center account, and **then** go to [https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq).

Find: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). 

Once there, select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

Hopes this helps.
 ~ *Mark*

closed 2021-05-04 by mhopkins-msft