MicrosoftDocs/windows-driver-docs #2768: Driver signed with signtool can't be installed in window2019 with test mode on

descartes123, 2021-04-19:
Our product has a ndis driver which followed the below doc to signed and installed for several years:
https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/tools-for-signing-drivers 

it mainly used below cmd to sign the driver:
makecert.exe -a sha256 -r -pe -n CN=$(DDKCertName) -sv $(DDKTestCertPath).pvk $(DDKTestCertPath).cer
inf2cat.exe /driver:$(DDKOutputDir) /os:ServerRS5_$(DDKInf2CatArch) /USELOCALTIME"
SignTool.exe sign /f $(DDKTestCertPath) /p <password> $(DDKOutputDir)\$(Name).$(Ext)"

And then call setup API to install driver to window.

It works well for all previous window version, but recently we need to upgrade to window 2019 which already opened the test mode,  the setup API always failed with error message "**A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.**"

I can't any solution from the official doc, does anyone know what wrong with it?

Thank in advance.

tedhudek, 2021-05-13:
Hi @descartes123, thanks for the feedback and apologies for the delay.

By test mode, did you enable testsigning in the BCD?

descartes123, 2021-05-14:
Of course we used Bcdedit.exe -set TESTSIGNING ON to enable test mode, we found it eventually copied the inf file to C:\Windows\System32\DriverStore folder and the driver binary to C:\Windows\System32\drivers folder after calling the INetCfgClassSetup->install method, but the driver service was not created, it just return 0x80070002, however,we can create the service manually by sc create <servicename> binPath="C:\Windows\System32\drivers\<drivername>.sys" type=kernel,and it could start well.
Not sure why this happened, no log or event trace could be watched, do you have any idea?
Thanks.

tedhudek, 2021-05-19:
That error is file not found, so something might be wrong in your INF.

If it's still a problem, you could try posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks for reaching out and best of luck with this issue.

closed 2021-05-19 by tedhudek