MicrosoftDocs/windows-driver-docs #2772: How do production signing options differ by Windows version?

GuidoKiener, 2021-04-20:
Your table in the question "How do production signing options differ by Windows version?" is confusing me.

Can you please confirm that a cross-signed driver is still valid for Windows 7, 8, and Server 2008 when it is signed with a nonexpired SHA-1 certificate (e.g. expires March 2023) and timestamped **after** July 1st 2021 and the same SHA-1 certificate chains to a cross-cert that is expired **before** July 1, 2021 (e.g. April 15, 2021).

AnAkkk, 2021-04-21:
What is written on the page seems to be very different from what Windows actually enforces. Multiple people have reported that expired cross-signing certificates are still working fine to load drivers on all versions of Windows. It seems like there aren't actually any checks on the expiration. Then maybe it will change with a future Windows update, who knows.

GuidoKiener, 2021-04-21:
Yes, it's true, we still can cross-sign drivers for Win 7 and Win 10 (when Secure Boot is disabled). I assume that the statement on the page is correct:
_"As long as driver packages are timestamped before the expiration date of the leaf signing certificate, they will continue working."_
I'm quite sure that there are checks on the expiration, which means you cannot timestamp your driver when your certificate is expired. Or if you omit timestamping and increment the time on your PC about 10 years then you can see that driver installation is refused.
Well, I just want to be sure that Microsoft does not have security updates that prevents loading of cross-signed drivers that are signed after July 1st, 2021. And I also expect that Microsoft will not change this policy on Windows 7 and 8. But this policy can quickly change when there is any bad cross-signed driver.

AnAkkk, 2021-04-21:
I can load a driver with an expired cross certificate and a timestamp just fine on Windows 10 (with Secure Boot disabled) - without changing the time or doing anything else. I have also tried to remove the timestamp parameter - it still loads fine without any issues.
So it doesn't look like there are currently any checks on the expiration of cross certificates.

GuidoKiener, 2021-04-21:
That's why I said driver installation is refused. Anything else are maybe flaws and we shoud not discuss it here ...

AnAkkk, 2021-07-02:
We're past the July 1st deadline, so I've tested again. Drivers are still loading just fine with expired cross certificates, including on Windows 11. Seems like the new policy has still not been implemented.

tedhudek, 2021-07-15:
Thanks for the feedback.

Please refer to https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates#how-do-production-signing-options-differ-by-windows-version for the latest guidance on signing options for different versions of Windows.

closed 2021-07-15 by tedhudek