MicrosoftDocs/windows-driver-docs #2774: Determining Whether a Leak Exists

syl1010, 2021-04-25:
This description is not clear. 
Where are the counters added - Data collector sets OR Performance Monitor? 
If the later how do you set the update time interval to 600 seconds

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 36af6504-4758-d4fd-def3-92bdecc0649d
* Version Independent ID: eb5e203b-89b6-0a76-7721-4366d978a5d9
* Content: [Determining Whether a Leak Exists - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/determining-whether-a-leak-exists)
* Content Source: [windows-driver-docs-pr/debugger/determining-whether-a-leak-exists.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/determining-whether-a-leak-exists.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-05-04:
@syl1010 - Thanks for the input on this topic. I added additional information to this topic to better describe how to right click on the Performance Monitor under Monitoring Tools and select Properties to set the time interval.  

Another tool that can be helpful in tracking memory usage trends is process explorer. https://docs.microsoft.com/en-us/sysinternals/downloads/process-explorer 
Good luck in tracking down the memory leak, and thanks for your interest in the Windows debugger documentation.

closed 2021-05-04 by DOMARS

syl1010, 2021-05-04:
Thank you. However, there is no option to see the "update time to 600 seconds" using Monitoring Tools->Performance Monitor.

These are the available options
[image]