MicrosoftDocs/windows-driver-docs #2776: invalid link for "EWDK for Windows 10, version 1903"

charliearcuri, 2021-04-26:
link to older EWDK download points at current EWDK download
**REPRO STEPS**
1) navigate to the "[Optional: Install the EWDK](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#optional-install-the-ewdk)" section of the "[Other WDK downloads](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)" page
2) observe link for older EWDK download:
  EWDK for Windows 10, version 1903

**EXPECTED**
It points point at page where a user can download previous version of EWDK.
**ACTUAL**
It points at the same pages as current EWDK.
    **EWDK for Windows 10, version 1903** -> _EWDK for Windows 10, version **2004** with Visual Studio Build Tools 16.7 (Recommended)_
I cannot get legacy EWDK version.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#optional-install-the-ewdk)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-04-26:
Hi @charliearcuri, thanks for the feedback.  I haven't tested it myself but noticed the two links at the bottom of the target page.  Did you try the second one there?

[image]

charliearcuri, 2021-04-26:
Thank you, Ted - that does unblock me - makes sense now (I think) - the EWDK for 1903 was released/validated with VS 16.0...  I'll give it a shot and circle back if that doesn't match what I'm looking for - thank you for the quick response.

Close this GitHub issue because I believe this does resolve the issue - user misunderstanding on my part. 😊👍

tedhudek, 2021-04-26:
Delighted to hear, Charlie!  It's far from intuitive, that's for sure.  Let me know if you have any issues going forward, happy to help.