MicrosoftDocs/windows-driver-docs #2779: few typo in code example

rajnishc8, 2021-04-27:
1. Typo issue 
Change following : 
Result =
  pIWbemServices->ExecMethod(
    ClassName,
    MethodName,
    0,
    NULL,
    &pInParameters,
 &pOutParameters,
    NULL
    );

To

Result =
  pIWbemServices->ExecMethod(
    ClassName,
    MethodName,
    0,
    NULL,
    pInParameters,       //change is in this line, note no & is required
 &pOutParameters,
    NULL
    );

2. Typo issue
Change following : 
Status = Parameter.ulval;
VariantClear(Parameter);

To
Status = Parameter.ulVal;
VariantClear(&Parameter);

Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 61c73017-14ce-7ec9-208f-95a162a64ba1
* Version Independent ID: bbb12c01-2f4d-1285-ecbc-7b7ab4ad6f73
* Content: [Injecting a Hardware Error - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/injecting-a-hardware-error)
* Content Source: [windows-driver-docs-pr/whea/injecting-a-hardware-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/injecting-a-hardware-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-04-27:
These changes look good, @rajnishc8, thanks for the report.  @mhopkins-msft will you please do the honors?

mhopkins-msft, 2021-04-28:
Hi @rajnishc8,

Thank you for reporting this. The topic has been updated.

 ~ *Mark*

closed 2021-04-28 by mhopkins-msft