MicrosoftDocs/windows-driver-docs #2790: Link to list of classes actually goes to list of classes you can't use

nathankidd, 2021-05-13:
The "system-defined device setup classes" link goes to a page titled "System-Defined Device Setup Classes Reserved for System Use" i.e. classes you can't use.  It would be really helpful if it linked to classes you _can_ use, as seems the intent.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f9dfcd91-a771-1c23-1dee-7c01f15051fc
* Version Independent ID: a990ee53-3120-3354-91eb-21404f2ba448
* Content: [Creating a New Device Setup Class - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/creating-a-new-device-setup-class)
* Content Source: [windows-driver-docs-pr/install/creating-a-new-device-setup-class.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/creating-a-new-device-setup-class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-05-13:
Hi @nathankidd, thanks for the feedback.  Does the following page help at all? https://docs.microsoft.com/en-us/windows-hardware/drivers/install/system-defined-device-setup-classes-available-to-vendors

nathankidd, 2021-05-13:
Yes, that is helpful and looks like the originally intended link.  Thanks.

tedhudek, 2021-05-13:
I updated the docs with the above commit - should appear on the live site within 5-10 minutes.

Thanks again for the heads-up!

closed 2021-05-13 by tedhudek