MicrosoftDocs/windows-driver-docs #2791: Is there an API for file signing LSA plugins and UEFI firmware?

kirillkovalenko, 2021-05-14:
There is  [Hardware dashboard API](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api) documentation. However, from examples provided it is not clear if and how the API can be used for [LSA/UEFI modules](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/lsa-and-uefi-file-signing). Please clarify.

mhopkins-msft, 2021-05-21:
Hi @kirillkovalenko,

I'm not really sure how to address your feedback. Do you have a specific task you're trying to accomplish for which you can't find documentation?

 ~ *Mark*

kirillkovalenko, 2021-05-21:
Hi @mhopkins-msft ,

> Do you have a specific task you're trying to accomplish for which you can't find documentation?

Yes, I have. I want to automate [LSA modules signing](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/lsa-and-uefi-file-signing) into our CI pipeline. I don't a documented API to accomplish it and currently I have to execute this task manually. However I see an API for automating very similar task,  i.e. [hardware drivers signing](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-submission). My guess was that LSA modules signing API actually exists but is not documented yet.

mhopkins-msft, 2021-05-24:
Hi @kirillkovalenko,

I checked with the dashboard development team and was told that there is currently no API for LSA signing.

Thanks,
 ~ *Mark*

closed 2021-05-24 by mhopkins-msft

BabaYaga1789, 2023-08-01:
Has the situation changed?