MicrosoftDocs/windows-driver-docs #2794: FSCTL_SET_PURGE_FAILURE_MODE missing from list of FSCTLs

rodwiddowson, 2021-05-18:
Lori (or whoever)

I have just spent a bunchy of time working out the impact of this FSCTL (its subtle for instance FltMgr doesn't respond correctly for cached writes and the status you return from an MJ_CREATE is not the one you return from a MJ_SET_INFO or MJ_WRITE)

It would have been incredibly handy had it been fully documented particularly given that the eviscerated FAT sources are insufficient.

I probably have enough information swapped in to put together a first draft of such a page, but you'd need to get the devs to check it.  However I don't want to waste my time if it is unlikely to be accepted.

Can you give me a steer on this:  Can you add this?  Should I try something or is this "internal use only"? 

Feel free to send me mail out of band if that's better

Rod
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 347e4909-0d3f-47c9-16f8-01b7d5ebb048
* Version Independent ID: a4bc09cb-8fdf-1dec-cf9f-e5dbc5161ba1
* Content: [IFS Reference - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/ifs-reference)
* Content Source: [windows-driver-docs-pr/ifs/ifs-reference.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/ifs-reference.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2021-05-18:
@rodwiddowson , the dev team and I would greatly appreciate you taking a crack at documenting FSCTL_SET_PURGE_FAILURE_MODE (sorry we missed doc'ing it in the first place). Feel free to submit a PR or you can email your draft to me directly, and I'll review with Christian. THANK YOU!

rodwiddowson, 2021-05-19:
I have entered a pretty ropy PR as #2796

closed 2021-05-19 by rodwiddowson