MicrosoftDocs/windows-driver-docs #2795: What prevents windows from using an audio driver?

philhuy, 2021-05-18:
What would cause windows to not recognize any compatible audio drivers after a windows update?  

Drivers have been reinstalled, and they show no issues in devmgmt.  
WIndows Audio and Endpoint services are running and showing no issues.

So, if the drivers are fine, and the audio services are fine, why does Windows believe there aren't any audio input/output devices?  Is there a breakdown in the audio engine somewhere?

Is there a way to delete the entire windows audio stack and restore it from an image?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0b65f03a-de45-b9db-381b-a9dff007c023
* Version Independent ID: 8676d46d-aeea-5809-d8e2-7651a14f709e
* Content: [Audio Devices Troubleshooting - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/audio-devices-troubleshooting)
* Content Source: [windows-driver-docs-pr/audio/audio-devices-troubleshooting.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/audio-devices-troubleshooting.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2021-05-22:
@philhuy . You should post your query in answers.microsoft.com .

DOMARS, 2021-05-24:
@philhuy -  This particular channel is monitored by the men and women who write the driver developer documentation posted on docs.microsoft.com/drivers. Though we'd like to, we don’t have the resources to provide support for specific scenarios. 

For audio driver issues, this article provides some information on audio troubleshooting and covers running the Audio Troubleshooting tool.  
https://answers.microsoft.com/en-us/windows/forum/windows_10-hardware/how-to-troubleshoot-sound-problems-in-windows-10/9807857d-632c-465a-ab10-395c4a1f6df4  

As Raju mentioned, you may also post your question on https://answers.microsoft.com

Good luck resolving your audio issues.

closed 2021-05-24 by DOMARS