MicrosoftDocs/windows-driver-docs #2800: More missing FSCTL documentation

rodwiddowson, 2021-05-21:
I am trying to make sense from a kernel point of view of what I am seeing when Win32 does a remote copy.  There is no driver documentation for these IOCTL/FSCTLS

[IOCTL_COPY_CHUNK](https://docs.microsoft.com/en-us/windows/win32/devnotes/ioctl-copychunk) the existing (user mode) documentation is probably sufficient for drivers.

[FSCTL_SRV_REQUEST_RESUME_KEY](https://docs.microsoft.com/en-us/windows/win32/devnotes/fsctl-srv-request-resume-key) is troubling

* It is defined in that page as `#define FSCTL_SRV_REQUEST_RESUME_KEY CTL_CODE(FILE_DEVICE_NETWORK_FILE_SYSTEM, 30, METHOD_BUFFERED, FILE_ANY_ACCESS)
` since it is not defined in any header
* Although it is entitled FSCTL the device code would indicate that this is an FILE system control and so will arrive down the `IRP_MJ_FILE_SYSTEM_CONTROL` call back.  This is certainly of importance to driver writers
* But we do see 0x140144 (as an **IOCTL**).  This would be `CTL_CODE(FILE_DEVICE_NETWORK_FILE_SYSTEM, 262, METHOD_BUFFERED,  FILE_ANY_ACCESS)` and seems to do much of what SRV_REQUEST_RESUME_KEY is documented to do

The user mode documentation for [IOCTL_LMR_DISABLE_LOCAL_BUFFERING](https://docs.microsoft.com/en-us/windows/win32/devnotes/ioctl-lmr-disable-local-buffering) does not give a CTL_CODE for it and I cannot find it in the 10.0.19041.0 WDK/SDK.  Procmon reports it by name so the definition must be somewhere.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 347e4909-0d3f-47c9-16f8-01b7d5ebb048
* Version Independent ID: a4bc09cb-8fdf-1dec-cf9f-e5dbc5161ba1
* Content: [IFS Reference - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/ifs-reference)
* Content Source: [windows-driver-docs-pr/ifs/ifs-reference.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/ifs-reference.md)
* Product: **windows-hardware**
* Technology: **ifs**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rodwiddowson, 2021-05-31:
@lorihollasch - my apologies - I just updated the initial comment since I had my FSCTLs and my IOCTLs confused

TL;DR `FSCTL_SRV_REQUEST_RESUME_KEY` is an IOCTL (not an FSCTL)

lorihollasch, 2021-09-10:
@rodwiddowson , I finally got a moment to look at this. None of the IOCTLs/FSCTLs that you mention are in a public WDK header file, so I can't document them. Unless I am looking with my eyes closed ... where are you seeing these? Assuming that I am looking correctly, I'm going to close this - feel free to reopen.

closed 2021-09-10 by lorihollasch