MicrosoftDocs/windows-driver-docs #2801: broken link

DSPatrick, 2021-05-25:
The Windows HLK, version 2004 link produces 
404 - File or directory not found.
The resource you are looking for might have been removed, had its name changed, or is temporarily unavailable.

https://docs.microsoft.com/en-us/windows-hardware/test/hlk/

EliotSeattle, 2021-05-25:
Hi @DSPatrick.

On the page https://docs.microsoft.com/en-us/windows-hardware/test/hlk/, the link to **Windows HLK, version 2004**, which points to [https://go.microsoft.com/fwlink/?linkid=2128655](https://go.microsoft.com/fwlink/?linkid=2128655), works for me, and it kicks off the download for the kit. 

Does that link still fail for you?

DSPatrick, 2021-05-25:
No, it appears to be working now. Thank you

EliotSeattle, 2021-05-25:
My work here is done!  :-)

Thanks!

closed 2021-05-25 by EliotSeattle

DSPatrick, 2021-05-25:
Yes, thank you!

Sent from Pixel 4