MicrosoftDocs/windows-driver-docs #2807: there is no Configuration Properties > Driver Install > Deployment

xspeed1989, 2021-05-27:
I have a KMDF driver project and a driver package in a solution, but  there is  no _Configuration Properties > Driver Install > Deployment_ 

My vs information:
[image]

tedhudek, 2021-05-27:
Thanks @xspeed1989 - can you please provide a link for the doc page you are referencing?

xspeed1989, 2021-05-28:
> Thanks @xspeed1989 - can you please provide a link for the doc page you are referencing?

@tedhudek 
doc page:https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-very-small-kmdf--driver

tedhudek, 2021-06-02:
@mhopkins-msft: looks like the above page needs at least one refreshed screenshot, similar to what you did for #2751

mhopkins-msft, 2021-06-03:
Hi @xspeed1989

Thank you for reporting this. I've updated the screenshots and directions to match Visual Studio 2019. The article should be republished later today.

 ~ *Mark*

closed 2021-06-03 by mhopkins-msft