MicrosoftDocs/windows-driver-docs #2811: Code Integrity Event Logs messages article needs an update

td47, 2021-05-31:
This article is dated 20 April 2017, and I suspect a lot has been added since then. At least this ID 3033 Event is happening a lot on my systems (here is a typical Event ID 3033):
Code Integrity determined that a process (\Device\HarddiskVolume1\Windows\System32\SIHClient.exe) attempted to load \Device\HarddiskVolume1\Program Files\Norton Security\Engine\22.20.5.39\symamsi.dll that did not meet the Windows signing level requirements.
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: beb749ad-f176-a1af-c95a-0c8f411ce523
* Version Independent ID: 78b75322-bdd4-1ac7-a41c-28a6e8651386
* Content: [Code Integrity Event Log Messages - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/code-integrity-event-log-messages)
* Content Source: [windows-driver-docs-pr/install/code-integrity-event-log-messages.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/code-integrity-event-log-messages.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-06-04:
Thanks for reaching out @td47.  Can you please clarify what changes you are proposing for the doc page in question?

td47, 2021-06-05:
@tedhudek - I am proposing that the document owner(s), gather together the error message changes (that must be a lot), that have been coded for, between Windows 1709 (Creators Update, or Redstone 3 as it is known internally, and the current date of the referenced document), and the CURRENT sofware levels, that NOW at 20H2.

I have already supplied ONE definate item not in your list (for the 3033 event), but if you want to get further examples, run the Event Viewer on a few systems that you have access to, and open the following Operational Log via this hierarchy:
>Application and Services Logs
           > Windows
                  > Code Integrity > Operational
Obviously the logs returned will depend upon any errors/problems seen with various drivers and AV executables and DLL objects, so resuts will vary. However, for Driver Developers, the error listing, if current, could result in MUCH more robust and resilient code, so that they can code for these conditions (and newer requirements), and do not fail over time, with the NEWER security measures that Microsoft adds frequently after CVE disclosures.

Given the huge amount of code changes and updates between 2017 and now, the document as it stands cannot possibly be current in its technical and detail content.

tedhudek, 2021-06-08:
Thanks for the clarification @td47.  After conferring with the team who owns this, I added a Note up top with a link to a more comprehensive page - which includes the 3033 error you mention.  Thanks again for the feedback!

closed 2021-06-08 by tedhudek