MicrosoftDocs/windows-driver-docs #2821: From Windows Driver Kit version 10.0.17740.0 I don't see "Device Type" option

sos82, 2021-06-17:
I see many guideline from Microsoft mention about "Device Type" in Device Config dialog. However, it is removed from version 10.0.17740.0. As I know it is available in version 10.0.17110.0 

Besides that, I can not make it works with Raspberry Pi. How to install "WDK Test Target Setup ARM-arm_en-us.msi" on Raspberry Pi?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1#feedback)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* Technology: **gettingstarted**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

mhopkins-msft, 2021-06-22:
#reassign:mhopkins-msft

mhopkins-msft, 2021-06-22:
Hi @sos82,

Thank you for reporting this. I've updated the screen shot to match the current version of Visual Studio 2019.

I'm sorry that I don't have the bandwidth to help you with installing the Raspberry Pi test target setup. You'll need to contact support on [support.microsoft.com](https://support.microsoft.com) for help with that. You could also post your question in Stack Overflow, [Microsoft Q&A](https://docs.microsoft.com/en-us/answers/products/), or the [OSR NTDEV forum](https://community.osr.com/categories/ntdev). 

Thanks again!
 ~ *Mark*

closed 2021-06-22 by mhopkins-msft