MicrosoftDocs/windows-driver-docs #2824: Downloading BinSkim as a ZIP file is not enough to build it

thiagopeixoto, 2021-06-23:
You need to clone the submodules, otherwise you'll get the following errors:

```
C:\binskim-main>BuildAndTest.cmd
Restoring packages...
C:\Program Files\dotnet\sdk\5.0.301\NuGet.targets(290,5): error MSB3202: O arquivo de projeto "C:\binskim-main\src\sarif-sdk\src\Sarif.Driver\Sarif.Driver.csproj" não foi encontrado. [C:\binskim-main\src\BinSkim.sln]
C:\Program Files\dotnet\sdk\5.0.301\NuGet.targets(290,5): error MSB3202: O arquivo de projeto "C:\binskim-main\src\sarif-sdk\src\Sarif\Sarif.csproj" não foi encontrado. [C:\binskim-main\src\BinSkim.sln]
Building solution...
Microsoft(R) Build Engine versão 16.10.1+2fd48ab73 para .NET
Copyright (C) Microsoft Corporation. Todos os direitos reservados.

C:\Program Files\dotnet\sdk\5.0.301\Sdks\Microsoft.NET.Sdk\targets\Microsoft.PackageDependencyResolution.targets(241,5): error NETSDK1004: Arquivo de ativos 'C:\binskim-main\src\BinSkim.Rules\obj\project.assets.json' não encontrado. Execute uma restauração de pacote do NuGet para gerar esse arquivo. [C:\binskim-main\src\BinSkim.Rules\BinSkim.Rules.csproj]
C:\Program Files\dotnet\sdk\5.0.301\Sdks\Microsoft.NET.Sdk\targets\Microsoft.PackageDependencyResolution.targets(241,5): error NETSDK1004: Arquivo de ativos 'C:\binskim-main\src\BinaryParsers\obj\project.assets.json' não encontrado. Execute uma restauração de pacote do NuGet para gerar esse arquivo. [C:\binskim-main\src\BinaryParsers\BinaryParsers.csproj]

FALHA da compilação.

C:\Program Files\dotnet\sdk\5.0.301\Sdks\Microsoft.NET.Sdk\targets\Microsoft.PackageDependencyResolution.targets(241,5): error NETSDK1004: Arquivo de ativos 'C:\binskim-main\src\BinSkim.Rules\obj\project.assets.json' não encontrado. Execute uma restauração de pacote do NuGet para gerar esse arquivo. [C:\binskim-main\src\BinSkim.Rules\BinSkim.Rules.csproj]
C:\Program Files\dotnet\sdk\5.0.301\Sdks\Microsoft.NET.Sdk\targets\Microsoft.PackageDependencyResolution.targets(241,5): error NETSDK1004: Arquivo de ativos 'C:\binskim-main\src\BinaryParsers\obj\project.assets.json' não encontrado. Execute uma restauração de pacote do NuGet para gerar esse arquivo. [C:\binskim-main\src\BinaryParsers\BinaryParsers.csproj]
    0 Aviso(s)
    2 Erro(s)

Tempo Decorrido 00:00:00.53
Build and test did not complete successfully.
```
Best regards.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6298d0a1-1f6b-6e00-5445-dfec501a4430
* Version Independent ID: 42f902a4-b1df-b863-7486-de26c54a205f
* Content: [Driver security checklist - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist)
* Content Source: [windows-driver-docs-pr/driversecurity/driver-security-checklist.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/driversecurity/driver-security-checklist.md)
* Product: **windows-hardware**
* Technology: **driversecurity**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-08-10:
@thiagopeixoto - Yes, I just downloaded the version 175 and it looks like it needs files from the sarif sdk project. So this is an issue.

https://github.com/microsoft/sarif-sdk/tree/main/src/Sarif.Driver

I attempted to add the package to the sarif project to BinSkim.sln file using the package manager in Visual Studio 2019 but no success.

Do you have the work around that worked for you handy?

```
Restoring packages for E:\Data1\Misc\Software\BinSkim175\binskim-main\binskim-main\src\BinSkim.Sdk\BinSkim.Sdk.csproj...
NU1104: Unable to find project 'E:\Data1\Misc\Software\BinSkim175\binskim-main\binskim-main\src\sarif-sdk\src\Sarif.Driver\Sarif.Driver.csproj'. Check that the project reference is valid and that the project file exists.
Package restore failed. Rolling back package changes for 'BinSkim.Sdk'.
```

I also see this issue in the BinSkim project related to simple directions.
https://github.com/microsoft/binskim/issues/211

I can follow up with the dev team, but if you have determined a work around, it would be interesting to learn about the approach you used that worked for you.

Thanks for filing this issue! Once we know the best path, here we will update the docs or make updates to the project build directions or files.

thiagopeixoto, 2021-08-13:
The easiest way i could find to compile it was using the `--recurse-submodules` option, which included the sarif sdk project.
```
git clone --recurse-submodules https://github.com/microsoft/binskim.git
Cloning into 'binskim'...
remote: Enumerating objects: 12386, done.
remote: Counting objects: 100% (2212/2212), done.
remote: Compressing objects: 100% (829/829), done.
remote: Total 12386 (delta 1797), reused 1634 (delta 1358), pack-reused 10174
Receiving objects: 100% (12386/12386), 43.97 MiB | 8.28 MiB/s, done.
Resolving deltas: 100% (10505/10505), done.
Updating files: 100% (1012/1012), done.
Submodule 'src/sarif-sdk' (https://github.com/microsoft/sarif-sdk) registered for path 'src/sarif-sdk'
Cloning into 'T:/Temp/binskim/src/sarif-sdk'...
remote: Enumerating objects: 48233, done.
remote: Counting objects: 100% (501/501), done.
remote: Compressing objects: 100% (271/271), done.
remote: Total 48233 (delta 281), reused 351 (delta 229), pack-reused 47732
Receiving objects: 100% (48233/48233), 102.27 MiB | 15.30 MiB/s, done.
Resolving deltas: 100% (39994/39994), done.
Submodule path 'src/sarif-sdk': checked out 'bfdd17782876d9b5002b7b73fef37a74aa748c9a'
```

I was also unable to add the package via the package manager in Visual Studio 2019.

DOMARS, 2021-08-13:
@thiagopeixoto - Thanks for that info! I was able to use the git recurse option to successfully build. It also looks like the dev team is adding to version 1.9 a check to download the sarif project. 

https://github.com/microsoft/binskim/commit/99012abb38b05e5aba7e4f61296b3b21d6bc0191

Will update the docs once that work is complete, and I re-test with the updated BuildAndTest.cmd.

thiagopeixoto, 2021-08-14:
That's great! :)

DOMARS, 2021-08-20:
@thiagopeixoto - I re-tested with the 1.9 updates and the recurse command seems to work great! So I updated the directions to use git clone, as that as it seems to be more in line with how the binskim project folks work with binskim. Big thanks for filing this issue, to save others the trouble of figuring out the work around.

closed 2021-08-20 by DOMARS