MicrosoftDocs/windows-driver-docs #2832: Windows.Devices.Printers hyperlink links to the incorrect winrt-version

zhit-microsoft, 2021-06-24:
Currently, the link https://docs.microsoft.com/en-us/uwp/api/windows.devices.printers will redirect to
https://docs.microsoft.com/en-us/uwp/api/windows.devices.printers?view=winrt-20348, which does not show the new API interfaces.

We should change the link to the following, so that the user can see the new interfaces without additional intervention.
https://docs.microsoft.com/en-us/uwp/api/windows.devices.printers?view=winrt-insider

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d38003a1-8fba-7709-b130-57493ec9475b
* Version Independent ID: 67f50a05-3916-244c-9eb4-3a259e79bc6d
* Content: [Print support app design guide - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devapps/print-support-app-design-guide?branch=master)
* Content Source: [windows-driver-docs-pr/devapps/print-support-app-design-guide.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devapps/print-support-app-design-guide.md)
* Product: **windows-hardware**
* Technology: **devapps**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

zhit-microsoft, 2021-06-24:
The same is true for the https://docs.microsoft.com/en-us/uwp/api/windows.graphics.printing.workflow hperlink.

RAJU2529, 2021-06-27:
@zhit-microsoft . Thanks for pointing out.

RAJU2529, 2021-07-01:
@barrygolden , Sir  Please check and merge this PR #2834

barrygolden, 2021-07-01:
@RAJU2529, thanks for submitting PR #2834

We're investigating this now and will update the issue when we have a resolution.

barrygolden, 2021-07-01:
Added "?view=winrt-insider&preserve-view=true" to six print support API namespace inks in two topics:

[Print support app design guide](https://docs.microsoft.com/windows-hardware/drivers/devapps/print-support-app-design-guide#see-also)

[Print support app association](https://docs.microsoft.com/windows-hardware/drivers/devapps/print-support-app-association#see-also)

barrygolden, 2021-07-01:
@zhit-microsoft, thanks for pointing out the incorrect redirects.

@RAJU2529, thanks again for submitting PR #2834.  The links have been updated per the comment above.

closed 2021-07-01 by barrygolden