MicrosoftDocs/windows-driver-docs #2844: Huh?

ksanger3838, 2021-07-09:
Clear as mud.  I searched for how to use signature keys.   Absolutely no info for a Key Novice to understand what to do.
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7659d22f-595a-6b02-566d-1325cf20d5cd
* Version Independent ID: 8dd845af-7408-99e7-20d0-331ad901c328
* Content: [Managing the Digital Signature or Code Signing Keys - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/managing-the-digital-signature-or-code-signing-keys)
* Content Source: [windows-driver-docs-pr/install/managing-the-digital-signature-or-code-signing-keys.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/managing-the-digital-signature-or-code-signing-keys.md)
* Product: **windows-hardware**
* Technology: **install**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-07-09:
Hi @ksanger3838, I agree, that page doesn't provide useful info. Can you give me a high level description of what you're trying to do with signature keys? I don't have much familiarity there myself but if I understood the scenario it might help me figure out who to ask :)

tedhudek, 2021-07-15:
Closing this one out for now, please reopen with more info if it's still important.

closed 2021-07-15 by tedhudek