MicrosoftDocs/windows-driver-docs #2845: link for MmGetSystemAddressForMdlSafe  is broken

rodwiddowson, 2021-07-10:
All the links for MmGetSystemAddressForMdlSafe point to `drivers/kernel/mm-bad-pointer` - and indeed the documentation for this function appears to have entirely disappeared,

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 29cf2a30-3b4d-4cec-33ff-1ed18e05c1e5
* Version Independent ID: 4c5683f8-157c-0a26-19d4-c65556e0e26b
* Content: [Using MDLs - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/using-mdls)
* Content Source: [windows-driver-docs-pr/kernel/using-mdls.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/using-mdls.md)
* Product: **windows-hardware**
* Technology: **kernel**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

rodwiddowson, 2021-07-10:
OIC that page is for all the kernel macros-  you need to thumb down it to find the one you care about.

So the info is there but buried.

I leave it to you to decide whether this matters but it confused me.  And, BTW, it stops <F1 in VisualStudio> and most search engines from finding it.

tedhudek, 2021-07-13:
It's a great point Rod, thanks for reminding me.  I think we can make some improvement here.  I'll keep this issue open for a bit to keep you apprised of progress.

RAJU2529, 2021-07-13:
@tedhudek , Sir i think the below link is perfect 
**https://docs.microsoft.com/windows-hardware/drivers/kernel/mm-bad-pointer#mmgetsystemaddressformdlsafe**

RAJU2529, 2021-07-13:
I am editing this article now.

tedhudek, 2021-07-16:
Hi @rodwiddowson,

I went ahead and creating new, standalone pages for many of the macros that had been documented on the mm-bad-pointer page.  I put links to the new pages in a See Also section at the bottom of https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/mm-bad-pointer (and a Note explaining this at the top).  

Also note the new TOC in the left pane when you are on that page.

Search engines should also be picking up the new pages as we speak, so hopefully soon you'll get these in search results (which I saw, as you did, that the old setup failed miserably on that score).

Really appreciate your opening this - it was of a tip of the iceberg sort of thing once I got into it!

And thanks to @RAJU2529 for the PR -- hopefully this overhaul will serve you much better.

Feedback welcome :)

closed 2021-07-16 by tedhudek

RAJU2529, 2021-07-17:
@tedhudek   Thanks for updating this article. Still exact location link must be update  .  I have found fee topic  links.
Witg your permission can i edit this article again.

RAJU2529, 2021-07-17:
@tedhudek . The following topics links are redirected to  this page **https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/mm-bad-pointer**

**
MmGetMdlVirtualAddress
MmGetMdlByteOffset
MmInitializeMdl
MmGetSystemAddressForMdlSafe**

tedhudek, 2021-07-19:
Thanks @RAJU2529 I am planning on fixing these links this week.  I'll keep you posted!

tedhudek, 2021-07-19:
Links should all be updated now @RAJU2529 - please test a couple and verify.  Thanks again to both of you for the sharp eye!

RAJU2529, 2021-07-20:
@tedhudek. all links are fixed. Thanks