MicrosoftDocs/windows-driver-docs #2865: Need new _OSI strings for OS releases past Win10 2004

rgough71, 2021-07-27:
Please update this page to define strings for releases beyond 2004 and Win 11.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7eea1fbb-7f7d-d32d-da26-07f6499b3336
* Version Independent ID: 3fed902a-fb51-cd76-29d1-3fdaa97d847f
* Content: [How to Identify the Windows Version in ACPI by Using _OSI - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/acpi/winacpi-osi)
* Content Source: [windows-driver-docs-pr/acpi/winacpi-osi.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/acpi/winacpi-osi.md)
* Product: **windows-hardware**
* Technology: **acpi**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-08-23:
Investigating next _OSI string entries for topic.

barrygolden, 2021-08-30:
Thanks for submitting this issue, @rgough71.  

I've checked in with the ACPI engineering team. The _OSI string for the next version of Windows is work in progress and has not yet been released to documentation for publication. I will update this topic as soon as I receive the information from the ACPI team.

barrygolden, 2021-08-31:
Closing this issue as no additional action is possible at this time. Will update the issue after the _OSI string for the next version of Windows has been added to this topic.

closed 2021-08-31 by barrygolden