MicrosoftDocs/windows-driver-docs #2867: Fix Buffer Descriptions

timrprobocom, 2021-07-29:
One of the things that most confuses driver developers especially is the usage of the two buffers in METHOD_IN_DIRECT and METHOD_OUT_DIRECT.  A small modification to this page could help that immensely.  Consider the text of this page:

```
For these transfer types, IRPs supply a pointer to a buffer at Irp->AssociatedIrp.SystemBuffer. This represents the buffer that is specified in calls to DeviceIoControl and IoBuildDeviceIoControlRequest. The buffer size is specified by Parameters.DeviceIoControl.InputBufferLength in the driver's IO_STACK_LOCATION structure.

For these transfer types, IRPs also supply a pointer to an MDL at Irp->MdlAddress. This represents the buffer that is specified in calls to DeviceIoControl and IoBuildDeviceIoControlRequest. This buffer can be used as either an input buffer or an output buffer, as follows:
```

Notice that BOTH descriptions say "This represents *the* buffer that is specified in calls...".  That's not helpful  You can't have two pointers to "the buffer".  This would be MUCH clearer if the first one said "This represents the first buffer" and the second one said "This represents the second buffer".  With that, it's crystal clear how these two pointers map to the parameters to `DeviceIoControl`.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4b8bfd5a-3ee5-e6c3-c77a-e80a20368690
* Version Independent ID: 1c7efd79-13fe-a5a1-49c9-30554d84d7ad
* Content: [Buffer Descriptions for I/O Control Codes - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/buffer-descriptions-for-i-o-control-codes)
* Content Source: [windows-driver-docs-pr/kernel/buffer-descriptions-for-i-o-control-codes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/buffer-descriptions-for-i-o-control-codes.md)
* Product: **windows-hardware**
* Technology: **kernel**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-07-30:
Hi @timrprobocom, thanks so much for the suggestion. That's an elegant way to do it. I made the change, and added some level 2 headings for the different methods, to make it easier to scan.  Change should be on the live site within 5 min or so - let me know what you think!

closed 2021-07-30 by tedhudek

timrprobocom, 2021-07-30:
Perfect!  This is one of the pages I come back to again and again, because after 30 years I still can't remember the IRP fields for the two buffers.