MicrosoftDocs/windows-driver-docs #2880: Only sources of EV certificates

jasonnet, 2021-08-20:
It would be helpful if the page indicated if the other companies advertising EV certificates and Authenticode EV certificates are recognized in the same way as the companies listed in the article.  Or if the list is simply the list of providers at the time the article was first published.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93d3f097-d81b-66c2-4835-d63758687d34
* Version Independent ID: fe9d1cdd-966e-89b5-c66b-60531b0ec655
* Content: [Get a code signing certificate - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate)
* Content Source: [windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2021-08-23:
Hi @jasonnet 

Thank you for this feedback. I've created a work item to research this for possible future updates.

Thanks,
 ~ *Mark*

closed 2021-08-23 by mhopkins-msft

jasonnet, 2021-08-25:
Thanks.  My English should have been better.   An abbreviated correction is: "...if the page indicated if THE CERTIFICATES OF the companies advertising... ...as THE CERTIFICATES OF the companies listed in the article".