MicrosoftDocs/windows-driver-docs #2882: Ambiguous "Important" Note

PeterGV, 2021-08-23:
In the "Important" note on this page, bullet point one is insufficiently clear.  It reads:

> "Extended validation (EV) code signing certificates are required for all submissions."

Does this mean to say "All submissions must be signed using an Extended Validation (EV) code signing certificate"?

It has long been true that an EV Cert has been a prerequisite to doing a submission.  It has been a feature, for which we here at OSR advocated long and loud, that -- having an EV Cert registered with the Dashboard -- a submission could be signed with ANY Class 3 Code Signing Certificate... the actual submission need not be signed with the EV Cert.

IF it's the intention of this "Important" note to notify folks that this policy has changed, it should probably be SUPER clear exactly what's required.

Hence my requested clarification.  I'm not sure this is the policy, so I didn't want to suggest the actual change via a PR.

Best Regards,

Peter

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9e0918f1-17f6-9f9c-30a7-47e75e14c257
* Version Independent ID: de7a4ab1-a17e-889a-16bf-d973023a2c88
* Content: [Add or Update a code signing certificate - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/update-a-code-signing-certificate)
* Content Source: [windows-driver-docs-pr/dashboard/update-a-code-signing-certificate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/update-a-code-signing-certificate.md)
* Product: **windows-hardware**
* Technology: **dashboard**
* GitHub Login: @mhopkins-msft
* Microsoft Alias: **mhopkins**

mhopkins-msft, 2021-09-08:
Hi Peter,

I'm still trying to get information to clarify this article. I want to make sure I'm representing your question correctly. I read it as either "You must have an EV Cert registered with the Dashboard. Once this is true, a submission can be signed with any class 3 code signing cert." OR it's "You must have an EV Cert registered with the Dashboard. In addition, each individual submission must be signed with the EV cert." Which is the most accurate way to put it?

Thanks,
 ~ *Mark*

PeterGV, 2021-09-08:
Hi Mark,

Option 1:
"You must have an EV Cert registered with the Dashboard. Once this is true, a submission can be signed with any class 3 code signing cert that has been previously registered with the Dashboard."

Option 2:
OR it's "You must have an EV Cert registered with the Dashboard. In addition, each individual submission must be signed with that same EV cert."

That's the question.  What IS the policy, and has it changed recently.

As of just a few months ago, and for the past several years, the MSFT policy was Option 2.  The doc page as written today implies, and that the policy has recently changed.

I was very​ involved, advocating for the Community. in getting Option 1 implemented as the policy several years back... It would be a Very Bad Thing if Option 2 were some new policy.  That'd really put companies with distributed development, test, and release organizations in a difficult situation.

Thanks for asking,

P

mhopkins-msft, 2021-09-08:
Ok cool, thanks for clarifying. I'll keep digging.

 ~ *Mark*

mhopkins-msft, 2021-09-10:
Hi Peter,

Thanks for your patience. I've updated the topic with the info I got from the engineering team. I appreciate you bringing this to my attention. Let me know if this doesn't resolve the issue.

Take care,
 ~ *Mark*

closed 2021-09-10 by mhopkins-msft